switchdoclabs/OurWeatherWeatherPlus #2: String.h not found on compile in Arduino IDE

YannChemin, 2017-04-21:
Hi,

I am using Arduino IDE, and it bugs on String.h, 
I have checked that Arduino.h is found in the same .ino file (few lines above).
Using Arduino 1.8.2 with the genericESP8266 board and the default compiler (AVRISP mkII).

Thanks for any suggestions,
Yann

switchdoclabs, 2017-04-21:
You need to make sure you have downloaded and selected the adafruit huzzah ESP8266 libraries

YannChemin, 2017-04-21:
I have not been able to find those libraries, 
only those related things: https://github.com/adafruit?utf8=%E2%9C%93&q=HUZZAH&type=&language=

any link plz

switchdoclabs, 2017-04-21:
Instructions:

https://www.adafruit.com/product/2471

YannChemin, 2017-04-21:
Yes I have included the ESP8266 boards listing and libraries already from the boards manager, and I selected that HUZZAH board (see previous message last line).
in Sketch-> Libraries -> I can find several ESP8266 things, but nothing mentioning HUZZAH.

switchdoclabs, 2017-04-21:
Did you try installing g the string.h library?

YannChemin, 2017-04-21:
You mean String.h ? or string.h ? the code looks for String.h

YannChemin, 2017-04-21:
by renaming String.h to string.h it moves on the next missing library (ellapsedMillis.h)

switchdoclabs, 2017-04-21:
String.h

switchdoclabs, 2017-04-21:
Did you follow the readme file?

It sound like you are missing a bunch of libraries.

YannChemin, 2017-04-21:
yeah it compiled after other libraries were added. it actually worked with string.h. from v17 to v21 now :-)

switchdoclabs, 2017-04-21:
:). '

closed 2018-09-28 by switchdoclabs