switchdoclabs/OurWeatherWeatherPlus #3: [Feature Request] Support for enterprise WiFi

ConorIA, 2017-04-24:
Hello, I am a PhD candidate at the University of Toronto. I bought a WeatherPlus station as a tinker project. Our plan is to run the station alongside our "professional grade" weather station (See: [weather.utsc.utoronto.ca](https://weather.utsc.utoronto.ca)).

Unfortunately, our campus WiFi is very restricted, and requires authentication either by private certificate, or by a authentication with a username and password (in the connection settings, not a captive portal). I have glanced through the WiFiManager source, and it seems to only support simple SSID and password combinations. 

I am opening this issue to track the possibility of more advanced WiFi configuration. In the meantime, I will also contact our networking folks to see if we can find a solution on our side.

switchdoclabs, 2017-04-24:
Sounds like a great project!!

That is correct.  You maybe able to add the certifications you need but it would be difficult.  

We suggest using a inexpensive wifi access point.  You should be able to get your network boys to sign off a limited use wifi network. 

SwitchDoc Labs

ConorIA, 2017-06-03:
Hi SwitchDoc team, a quick follow-up. One of our network guys suggested we set up a 5GHz WiFi with an old  Point-to-Point AP that we have on the building roof. Can the WeatherPlus board connect to 5.0 GHz? I'm in the field at the moment and don't have access to test the station myself just now.

Strykar, 2018-01-17:
@ConorIA Unfortunately no, the Espressif chip can't do 802.11a (5 Ghz) nor RADIUS (certificates). Check out https://github.com/esp8266/esp8266-wiki/wiki
Get an inexpensive _simultaneous_ dual-band radio like http://www.tp-link.com/us/products/details/cat-5506_Archer-C7.html and connect the WeatherPlus to the 2.4 Ghz network and set the Univ. network as the gateway/internet via 5.8 Ghz on the router.