switchdoclabs/OurWeatherWeatherPlus #16: Unresolved includes

gb0101010101, 2019-03-20:
There are a number of includes that fail to resolve when building on linux.

I have installed ESP8266 boards from http://arduino.esp8266.com/stable/package_esp8266com_index.json but could only find "Adafruit Feather HUZZAH ESP8266" and not "Adafruit HUZZAH ESP8266" as stated in BUILD.MD. Is this the correct board?

Directions from Adafruit say to use this board: https://learn.adafruit.com/adafruit-huzzah-esp8266-breakout/using-arduino-ide

'#include "Wire.h"' used in multiple libraries point to local file that does not exist. Is this supposed to be '#include <Wire.h>' which would use the file from the ESP8266 library?

'#include "Crc16.h"' in SDL_ESP8266_WeatherPlus.ino. Local file is lowercase 'crc16.h' and therefore fails to include on Case Sensitive Linux OS. Renaming file solves the problem. Same goes for '#include "elapsedMillis.h"' in same INO file. 

Conversely there is '#include <String.h>' which should be lowercase '#include <string.h>'

Then it gets a bit more complex. There is '#include <Time.h>'  in SDL_Weather_80422.cpp. Is this supposed to use the local file 'TimeLib.h' or  '#include <time.h>' or 'Time' library by Michael Margolis?

Finally there is '#include "user_interface.h"' in local WifiManager.h. "user_interface.h" is not in your repo.

I see that others have brought up similar issues but they were not fixed in repo:
https://github.com/switchdoclabs/OurWeatherWeatherPlus/issues/2
https://github.com/switchdoclabs/OurWeatherWeatherPlus/issues/4
https://github.com/switchdoclabs/OurWeatherWeatherPlus/issues/14

Thanks for any info you provide. Should be receiving my kit in 2 days and would like to get the code compiling before then.

switchdoclabs, 2019-03-20:
Hi,

We usually answer these questions over on the Forum at forum.switchdoc.com <http://forum.switchdoc.com/>

What IDE are you using to build with?

Here are some answers partially at least:

> There are a number of includes that fail to resolve when building on linux.
>
> I have installed ESP8266 boards from http://arduino.esp8266.com/stable/package_esp8266com_index.json [...]
Yes

> Directions from Adafruit say to use this board: https://learn.adafruit.com/adafruit-huzzah-esp8266-breakout/using-arduino-ide [...]

I’d have to say that I’m not sure.  We compile it using the Arduino IDE and it is resolved correctly.  If it compiles the other way, then you are probably right.   Note:  Libraries and the Arduino IDE is one of the big problem with moving distributions from one place to another as far as IDEs go.

> '#include "Crc16.h"' in SDL_ESP8266_WeatherPlus.ino. Local file is lowercase 'crc16.h' and therefore fails to include on Case Sensitive Linux OS. Renaming file solves the problem. Same goes for '#include "elapsedMillis.h"' in same INO file.
Yes, that makes sense.
> Conversely there is '#include <String.h>' which should be lowercase '#include <string.h>'
Same as above.
> Then it gets a bit more complex. There is '#include <Time.h>' in SDL_Weather_80422.cpp. Is this supposed to use the local file 'TimeLib.h' or '#include <time.h>' or 'Time' library by Michael Margolis?
This is a real problem.   I believe TimeLib.h is correct, but again it depends.  Oh, that is a bit of a pun.
> Finally there is '#include "user_interface.h"' in local WifiManager.h. "user_interface.h" is not in your repo.
OK, I don’t have that repo here to check.  I can check it later at the lab.  Again, we use the Arduino IDE on Mac and Windows to compile.  Even if I am a Linux guy.

Best regards,

John Shovic

gb0101010101, 2019-03-21:
Hi John,

Thanks for the prompt reply. I managed to get it to compile with the following changes:

#include "Wire.h"    ->    #include <Wire.h>
#include "Crc16.h"  ->    Renamed file from repo to match
#include "elapsedMillis.h"    ->    Renamed file from repo to match
#include <String.h>    ->     #include <string.h>
#include <Time.h>    ->     #include <time.h>

#include "user_interface.h" problem disappeared without me changing anything. No I dea why it happened.

Also had to modify 2 occurrences of Wire.requestFrom() in WXLink.h
Wire.requestFrom(0x08, 32, true);    ->     Wire.requestFrom(8, 32, 1);
This is using ESP8266 Wire v2.4.2 library. I think the 2nd value was wrong type but chnaged all to 'int' to match candidate for an easier change.

Will post other issues on the Switchdoc Forum of which hopefully there will be none. Just wanted to respond here so that others would know what needed to be changed.

ConorIA, 2020-04-08:
Thank you @gb0101010101 for this guide. I followed your indications in https://github.com/switchdoclabs/OurWeatherWeatherPlus/issues/16#issuecomment-475109321 and managed to compile. The changes should be made in the repo. I may do my part and open a PR later in the week. Also, I found that `WifiManager.h` needs to be updated for newer versions of the https://github.com/esp8266/Arduino. (I was running into [this bug](https://arduino.stackexchange.com/questions/72581/esp8266-compilation-error-previous-declaration-of-httpmethod-http-head)) Instead of taking on that challenge, I just installed version 2.5.0 (I tried to guess what version may have been used to compile here) and that worked.

MrDornback, 2020-04-30:
A word of advice, which bit me in my backside when I wasn't paying attention.
If you change includes from local "" to global <>, you may be missing some custom functions/handlers that are specific to the SDL software and .h .cpp files.

I found this when dealing with an HTTP_HEAD re-declaration issue...  
[https://github.com/switchdoclabs/OurWeatherWeatherPlus/pull/23](https://github.com/switchdoclabs/OurWeatherWeatherPlus/pull/23)